r45635/HVAC-IR-Control #2: Mitsubishi MSY-GE10VA

bilogic, 2015-12-28:
Hi,

I tried this code, but no reaction from the AC. 
I have confirmed that the IR LED is indeed blinking (using smartphone camera)

Below are the RAW codes for my on and off. Any tips on the best way to get started? Thank you.

```
//aircon  ON Raw: (583) 3436, -1724, 440, -1284, 408, -1316, 444, -424, 440, -424, 440, -428, 436, -1288, 436, -428, 440, -424, 412, -1316, 440, -1280, 436, -436, 440, -1284, 440, -428, 436, -424, 412, -1316, 444, -1280, 436, -428, 440, -1280, 416, -1308, 440, -428, 408, -460, 412, -1316, 436, -424, 412, -456, 440, -1288, 412, -456, 412, -456, 412, -456, 436, -432, 412, -460, 412, -456, 440, -428, 440, -428, 440, -424, 436, -432, 412, -460, 436, -424, 440, -424, 440, -436, 436, -428, 440, -432, 436, -428, 440, -424, 440, -432, 436, -428, 440, -1280, 412, -456, 440, -428, 436, -432, 412, -452, 440, -428, 440, -1284, 440, -1276, 468, -412, 436, -424, 436, -444, 408, -460, 440, -1288, 412, -452, 440, -1284, 440, -436, 412, -460, 440, -424, 412, -456, 436, -428, 468, -1264, 412, -1316, 440, -424, 468, -1264, 412, -1312, 408, -456, 440, -432, 412, -460, 436, -424, 440, -428, 444, -420, 412, -456, 440, -428, 436, -1288, 440, -424, 440, -1280, 440, -1280, 440, -432, 408, -460, 440, -428, 440, -424, 412, -456, 412, -456, 412, -456, 412, -456, 412, -456, 412, -456, 412, -460, 432, -440, 408, -460, 436, -432, 436, -428, 440, -424, 440, -428, 412, -460, 440, -424, 440, -420, 440, -440, 436, -424, 444, -428, 444, -420, 440, -424, 444, -428, 412, -452, 412, -456, 436, -440, 436, -432, 440, -428, 412, -460, 436, -428, 412, -452, 440, -428, 440, -1284, 412, -456, 412, -456, 440, -424, 440, -428, 440, -420, 408, -456, 412, -460, 436, -424, 412, -456, 440, -432, 436, -432, 440, -424, 412, -456, 440, -424, 412, -456, 412, -456, 408, -456, 440, -424, 412, -456, 440, -428, 440, -424, 412, -456, 412, -1312, 440, -1292, 416, -1308, 412, -1316, 440, -17200, 3512, -1744, 444, -1280, 412, -1312, 440, -428, 436, -428, 444, -424, 440, -1280, 440, -424, 444, -424, 408, -1316, 440, -1280, 440, -424, 436, -1288, 440, -424, 408, -464, 408, -1312, 436, -1288, 440, -432, 408, -1316, 440, -1288, 440, -432, 440, -428, 440, -1280, 436, -428, 440, -428, 436, -1284, 440, -424, 440, -424, 440, -428, 440, -428, 440, -424, 440, -424, 440, -424, 476, -404, 440, -424, 412, -464, 436, -424, 440, -424, 412, -456, 412, -456, 436, -428, 436, -432, 440, -424, 436, -428, 436, -428, 436, -432, 440, -1288, 412, -448, 436, -436, 440, -424, 408, -460, 436, -432, 436, -1292, 412, -1316, 412, -452, 408, -456, 424, -444, 436, -432, 440, -1284, 440, -420, 412, -1320, 436, -424, 412, -456, 436, -428, 412, -452, 436, -436, 440, -1284, 440, -1280, 412, -460, 440, -1280, 412, -1316, 412, -456, 436, -428, 440, -428, 440, -428, 440, -424, 440, -428, 412, -460, 436, -428, 412, -1316, 436, -432, 436, -1280, 436, -1284, 444, -424, 408, -460, 412, -484, 412, -428, 436, -428, 444, -420, 444, -420, 412, -456, 412, -456, 412, -456, 412, -456, 412, -464, 412, -456, 440, -428, 440, -424, 440, -424, 440, -432, 408, -460, 440, -424, 440, -420, 444, -436, 440, -424, 440, -428, 440, -424, 440, -428, 440, -428, 412, -456, 408, -460, 436, -436, 440, -428, 440, -428, 412, -460, 440, -424, 412, -452, 416, -452, 440, -1284, 436, -428, 436, -428, 440, -428, 412, -456, 440, -428, 412, -460, 440, -424, 440, -428, 436, -428, 440, -424, 436, -432, 408, -464, 440, -424, 412, -460, 436, -432, 468, -408, 436, -428, 412, -464, 444, -424, 412, -452, 440, -428, 440, -428, 440, -1276, 412, -1316, 440, -1284, 444, -1276, 440, 

//aircon OFF Raw: (583) 3412, -1748, 416, -1312, 436, -1284, 408, -460, 436, -428, 408, -464, 412, -1308, 412, -456, 412, -456, 436, -1284, 412, -1312, 440, -424, 408, -1316, 440, -424, 408, -460, 412, -1312, 412, -1316, 440, -428, 412, -1312, 412, -1312, 412, -456, 444, -424, 440, -1276, 416, -464, 440, -424, 440, -1284, 440, -424, 408, -460, 408, -460, 436, -428, 412, -460, 412, -452, 408, -464, 408, -456, 412, -456, 412, -448, 412, -456, 408, -456, 412, -456, 440, -424, 408, -460, 408, -456, 412, -456, 440, -428, 440, -420, 412, -456, 476, -404, 440, -424, 408, -468, 412, -456, 440, -424, 440, -428, 440, -1280, 440, -1284, 412, -456, 440, -428, 440, -428, 436, -432, 412, -1312, 412, -456, 440, -1284, 412, -456, 412, -452, 412, -456, 412, -456, 444, -420, 440, -1284, 408, -1320, 412, -452, 412, -1312, 440, -1284, 412, -452, 436, -428, 436, -436, 412, -452, 436, -436, 436, -428, 440, -424, 408, -460, 440, -1280, 412, -460, 412, -1316, 412, -1316, 408, -456, 412, -452, 412, -456, 408, -460, 408, -456, 440, -428, 412, -456, 440, -424, 412, -456, 412, -460, 408, -460, 472, -404, 436, -428, 408, -468, 412, -460, 436, -428, 440, -424, 440, -424, 412, -456, 412, -456, 408, -456, 412, -452, 412, -460, 420, -444, 412, -452, 412, -456, 432, -444, 412, -460, 440, -424, 408, -456, 444, -424, 412, -452, 436, -432, 440, -424, 412, -460, 412, -1312, 412, -456, 412, -456, 444, -420, 436, -432, 412, -460, 440, -424, 412, -456, 408, -460, 472, -408, 412, -452, 412, -456, 444, -420, 412, -452, 436, -432, 412, -456, 444, -424, 408, -456, 412, -456, 436, -428, 408, -460, 408, -460, 412, -456, 432, -1292, 412, -456, 468, -1264, 412, -1312, 412, -17096, 3536, -1724, 472, -1264, 412, -1312, 412, -456, 408, -456, 408, -460, 412, -1312, 412, -452, 412, -456, 436, -1284, 412, -1312, 408, -456, 412, -1308, 440, -432, 412, -452, 444, -1280, 412, -1316, 440, -428, 436, -1284, 416, -1312, 412, -456, 412, -456, 412, -1316, 440, -428, 440, -428, 408, -1312, 412, -456, 440, -424, 440, -428, 440, -424, 440, -424, 408, -468, 440, -424, 436, -432, 440, -424, 440, -428, 440, -428, 412, -456, 408, -456, 436, -440, 440, -428, 436, -432, 412, -460, 440, -424, 408, -460, 436, -428, 412, -452, 436, -432, 436, -428, 440, -424, 408, -456, 412, -456, 436, -1284, 412, -1312, 412, -456, 440, -424, 412, -456, 440, -424, 436, -1288, 412, -456, 412, -1312, 408, -460, 408, -456, 440, -424, 440, -428, 412, -460, 412, -1312, 440, -1284, 412, -456, 436, -1284, 412, -1312, 440, -428, 408, -460, 440, -420, 440, -428, 436, -432, 412, -452, 412, -456, 408, -456, 408, -1316, 436, -424, 412, -1316, 436, -1284, 440, -420, 440, -428, 436, -424, 412, -452, 412, -460, 412, -456, 440, -424, 440, -424, 440, -428, 440, -428, 440, -424, 440, -424, 412, -456, 440, -428, 412, -452, 412, -464, 436, -424, 412, -456, 440, -428, 436, -424, 440, -428, 440, -424, 440, -424, 412, -456, 408, -456, 440, -424, 440, -432, 412, -460, 440, -424, 412, -452, 412, -456, 464, -416, 436, -436, 412, -452, 440, -428, 408, -1316, 440, -432, 440, -428, 436, -428, 440, -428, 408, -460, 440, -424, 436, -432, 440, -424, 440, -424, 440, -428, 440, -420, 468, -408, 436, -436, 408, -456, 436, -436, 412, -452, 440, -428, 436, -432, 436, -428, 436, -428, 436, -428, 416, -452, 412, -1312, 412, -456, 436, -1288, 440, -1284, 412, 
```

cosmopaco, 2015-12-29:
you could try?https://github.com/ToniA/arduino-heatpumpir

r45635, 2015-12-30:
Hello, in order to help, could you let me know from where is coming your raw code ? Are you using the the sample demo included  ?
regards,

bilogic, 2015-12-30:
Hi,

My raw code was captured using the AnalysIR sketch found here:
http://www.analysir.com/blog/wp-content/uploads/2014/03/Arduino_Record_Long_AirConditioner_Infrared_Signals_10.txt

The sample I tested was https://github.com/r45635/HVAC-IR-Control/blob/master/HVACDemo/HVACDemo.ino, there was no reaction from the AC.

r45635, 2015-12-30:
Ok for clarification. 
At that point i'm not able to use your raw code. As far i remind the soft from Chris (Analysis IR) should be able to detect your Mitsubishi protocol and therefore should provide you binary data instead of raw. 
I would strongly suggest to use 2 arduino for your debug purpose. Use one in reception with AnalysisIR and another one just for emission. 

some basic checks : make sure to have an IR led emitter, make sure IR led is pointed directly to your HVAC (mine are very sensitive if you're not emitting to his direction). 

regards,

bt4wang, 2018-03-22:
Hi @bilogic . I've checked your IR code, there is a tiny difference between your code and @r45635 's protocol. The third bit of Byte 14 in your code is 1, which was not informed by @r45635 .
[image]

I think you can try this:

edit IRremote2.cpp, at line 399

`byte data[18] = { 0x23, 0xCB, 0x26, 0x01, 0x00, 0x20, 0x08, 0x06, 0x30, 0x45, 0x67, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1F };`

change the 15th(aka Byte 14) HEX number to 0x20, so the line will be like this

`byte data[18] = { 0x23, 0xCB, 0x26, 0x01, 0x00, 0x20, 0x08, 0x06, 0x30, 0x45, 0x67, 0x00, 0x00, 0x00, 0x20, 0x00, 0x00, 0x1F };`

if this doesn't work, I think you should check your IR emitter, use another arduino board to recieve the code your sent and have it checked.

r45635, 2019-01-01:
Thanks @bt4wang, i propose to closed that issue waiting a feedback confirmation from someone if such bit value is the reason of no HVAC reaction for this model.

closed 2019-01-01 by r45635

bilogic, 2019-01-01:
Hi, 

https://github.com/ToniA/arduino-heatpumpir got it working for me, his repo supports the P/N KM09G 0116667 remote control. Thank you.

Ericmas001, 2019-01-01:
I've read the code and I think the only difference is that when sending the "twin" packets, for msy he changes one byte at position 14.
```
    // Pause between the first and the second data burst

    // Also modify one byte for the second burst on MSY model. This does not affect the checksum of the second burst

    if (j == 0) {

      IR.mark(MITSUBISHI_AIRCON1_BIT_MARK);

      IR.space(MITSUBISHI_AIRCON1_MSG_SPACE);

      if (_mitsubishiModel == MITSUBISHI_MSY) {

        MitsubishiTemplate[14] = 0x24;

      }

    }
```