r45635/HVAC-IR-Control #4: Documentation

Dustyfrogz, 2016-01-14:
Salut,
Super boulot!!!
Etant débutant y a t'il un site où l'on peut trouver plus d'explication?
J'ai une clim mitsubishi, totaline et samsung si je peux aider.

Mat2Vence, 2016-01-14:
Bonjour,
Quelles types d'explications attends-tu? Concernant la partie reverse engineering, elle a été réalisée à l'aide du logiciel AnalysIR qui permet de simplifier pas mal l'analyse des trames des télécommandes.

Dustyfrogz, 2016-01-14:
Je n'ai pas AnalysisIR. 
Je suis arrivé ici car j'ai copié les codes de ma telecommande avec IRremote. 
La clim totaline c'est du nec et mitsu apres avoir agrandi la taille du buf j'ai les raws.
Mais impossible d'envoyer avec IRremote la data de la clim totaline. Comme pour la mitsu je suis obligé d'envoyer les raws pour que la clim réagisse.
J'ai essayé ton code HVACdemo mais rien ne se passe, en meme temps je sais pas si l'envoi est automatique. J'utilise une Nano donc pin3 pour l'emetteur?

Mat2Vence, 2016-02-03:
Le truc c'est que selon le protocole exact de ton clim, tu vas collé ou non avec le code.
En effet, le code tel que développé est issue d'un reverse engineering sur les trames en RAW de telle sorte qu'on ait juste besoin de passer des paramètre intelligible (température, mode, FAN...). Mais cette transposition est propre à chaque protocole. Même sur une même marque d'appareil, je ne suis pas certain que les protocole soit systématiquement les mêmes.

marcosemiao, 2016-02-10:
En effet, je possède une clim inverter et donc j'ai repris le code.
Quand j'ai testé il y avait certaine commande qui fonctionne pas.

Allumer et éteindre, changer la température, cela fonctionnait bien.
Mais par exemple, le mode HVAC_HOT, HVAC_DRY et HVAC_AUTO fonctionnait mais le mode HVAC_COLD ne fonctionnait pas.

il y a un décalage de bit entre le code fournit ici et ma clim.

dans le code on a : B00011000 alors que moi j'ai B00110000

wixaw, 2019-03-14:
Bonjour,
Je débute avec RPI et ESP, je voudrais faire fonctionner ceci avec ESPEASY.
Et effectivement comme le titre du ticket j'aimerais une documentation de mise en place.
J'ai un IR émetteur et un IR récepteur, comment puis-je faire  ? 
Merci 
William