r45635/HVAC-IR-Control #8: Cannot Upload Library to Arduino

janvikalra, 2016-07-08:
As it stands I cannot add this master file into Arduino and thus I cannot use the Mitsubishi Send sketches that I seek.

I first tried to add this file into my Arduino by Arduino > Sketch > Include Library > Add .Zip File but it says "Specified folder/zip file does not contain a valid library" 

I then tried to add it manually by upzipping the file, taking out all folders and putting it in the library folder. However, when I then try to include the library by Arduino > Sketch > Include Library only the HVACDemo folder comes up.

Where am I going wrong?

Mat2Vence, 2016-07-08:
As far as I remember, we did not switch the sketch into a library. It means then you do not haev to import it as a library but just use it as you main sketch and make any modification you need for your system to work as you expect.
Hope it helps

janvikalra, 2016-07-11:
Awesome - thank you!

closed 2016-07-11 by janvikalra