r45635/HVAC-IR-Control #10: does not compile

9H1LO, 2016-08-14:
actesting.ino:29:3: error: variable or field 'sendHvacMitsubishi' declared void
actesting.ino:29:3: error: 'HvacMode' was not declared in this scope
actesting.ino:30:3: error: expected primary-expression before 'int'
actesting.ino:31:3: error: 'HvacFanMode' was not declared in this scope
actesting.ino:32:3: error: 'HvacVanneMode' was not declared in this scope
actesting.ino:33:3: error: expected primary-expression before 'int'
variable or field 'sendHvacMitsubishi' declared void

wibbly, 2016-08-29:
I'm having the same trouble with the code at:  HVAC-IR-Control/HVAC_ESP8266/HVAC_ESP8266T.ino

toshiba_esp8266:25: error: variable or field 'sendHvacToshiba' declared void
toshiba_esp8266:25: error: 'HvacMode' was not declared in this scope
toshiba_esp8266:26: error: expected primary-expression before 'int'
toshiba_esp8266:27: error: 'HvacFanMode' was not declared in this scope
toshiba_esp8266:28: error: expected primary-expression before 'int'
variable or field 'sendHvacToshiba' declared void

NicksonYap, 2017-04-04:
Fork from: https://github.com/susumuota/HVAC-IR-Control
Compiles but doesn't seem to work for my Panasonic air cond.
No response from the air cond.