r45635/HVAC-IR-Control #12: Compile error on Arduino Uno

M-ax-G, 2017-01-12:
When I compile an unmodified HVACDemo.ino with IRremote2.h, I get a compile error, which I could not (re)solve:

D:\Temp\W7Temp\ccRC7vA1.ltrans0.ltrans.o: In function `loop':

D:\MaxG_MyDocuments\Arduino\zTests\HVAC_IR/HVAC_IR.ino:44: undefined reference to `IRsend::sendHvacMitsubishi(HvacMode, int, HvacFanMode, HvacVanneMode, int)'

D:\MaxG_MyDocuments\Arduino\zTests\HVAC_IR/HVAC_IR.ino:48: undefined reference to `IRsend::sendHvacMitsubishi(HvacMode, int, HvacFanMode, HvacVanneMode, int)'

D:\MaxG_MyDocuments\Arduino\zTests\HVAC_IR/HVAC_IR.ino:54: undefined reference to `IRsend::sendHvacMitsubishi(HvacMode, int, HvacFanMode, HvacVanneMode, int)'

D:\MaxG_MyDocuments\Arduino\zTests\HVAC_IR/HVAC_IR.ino:58: undefined reference to `IRsend::sendHvacMitsubishi(HvacMode, int, HvacFanMode, HvacVanneMode, int)'

D:\MaxG_MyDocuments\Arduino\zTests\HVAC_IR/HVAC_IR.ino:62: undefined reference to `IRsend::sendHvacMitsubishi(HvacMode, int, HvacFanMode, HvacVanneMode, int)'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling.

M-ax-G, 2017-01-12:
I have to add: the library IRremote2.h has been installed properly, and even included in the .ino, just to prove the point. I have created the irsend() call from scratch,which then looks like the ones in the .ino code and looks correct to me. But why does it through the error?

M-ax-G, 2017-01-13:
There a quite a few errors in the .cpp and.h files; once I fixed these, it compiles...

closed 2017-01-13 by M-ax-G

micktech, 2017-01-14:
If you don't mind me asking, what changes did you have to make M-ax-G ? Having a few issues compiling it myself. Regards.

M-ax-G, 2017-01-14:
OK... I had no time to look at the inner workings of the code, thus, have yet no grasp on how it all fits together. In this instance I am only interested in the Mitsubishi A/C... and just ran a quick compare.

The changes in IRremote.ccp:
remove slash before comment in lines: 221, 222, 385, 386, 510, 511, 652, 653
add slash in line: 651
remove closing semicolon in line 663
commented out line 665 - variable is not used
commented out line 639 to including line 763, because variables in line 739 and 740 are not defined, and I do not need the Toshiba functionality.

changes in IRremoteInt2.h
commented out line 153: defied twice; may need to be renamed to Toshiba (if it needs to exist there)

This should work... as in it does compile now without errors; still have to hook up an IR LED and test it. :)