r45635/HVAC-IR-Control #14: Licence/copyright question?

crankyoldgit, 2017-02-27:
Hi, I've had a user of our library [request](https://github.com/markszabo/IRremoteESP8266/issues/70) us to add support for Mitsubishi A/C remotes. They've pointed us at your repo. As noted in #1 there is no licence set for this code. There is a copyright notice on some of the files, hence we can't touch it without a licence or explicit permission etc.

Any chance you can add a suitable (open pref) license, or give us explicit permission?

Ping @r45635 as it's their name on the copyright notice.

r45635, 2017-02-27:
Hi, ack. too long i did not tackle this point. I will do this ASAP and confirm this to you.
to avoid any doubt this is personnal works then copyright is simple; for the license rights : it will be naturally suitable for the community needs.
i'll come back when done.
regards,
r45635

r45635, 2017-02-27:
License.txt added.

closed 2017-02-27 by r45635

crankyoldgit, 2017-02-27:
Thanks. Just to clarify/confirm, as IRremoteESP8266 is under LGPL 2.1 and you've chosen GPL3.
LGPL(clause 14) suggests we get explicit permission to license the bits we may use under this licence.
I'll probably have to use a separate file to cover any code under GPL3. Of course, all attribution will be added and links back to his repo if we use it. It's probably overkill to ask/check/do this (IANAL), but just playing it safe.

So, are you happy with/agree to that?