r45635/HVAC-IR-Control #20: Not an issue as such, how to use on Raspi..

35nine, 2018-04-03:
Hi.  Am banging my head on a wall here, have attempted to use the python port but no ir activity seen from ir led used.  I have pigpio installed, have used the pip install command from the readme and then am lost, I admit not to know how to use this within python.  I have run the mits demo code, it throws up no errors as such, just no ir light emitted.  IR led is known to be fine, lirc commands all fine.  If anyone can offer some extra guidance how to use, would be fantastic.  Thanks

Ericmas001, 2018-04-07:
Hi !
I personnally use it everyday with a rpi-zero-w with Minibian installed.

The important thing is:
When you use "hvac = Mitsubishi(23, LogLevel.ErrorsOnly)", 23 is the GPIO number, not the PIN number
My IR led is on GPIO 23 (Pin 16). 
[image]

I will give you more info than you need, so you can reproduce something close to my setup.
This is my setup:
https://github.com/Ericmas001/pi-adventures/blob/master/general/minibian_initial_setup.md

Then I installed:
```
sudo apt-get install git -y
sudo apt-get install python python-dev python-pip python-rpi.gpio pigpio -y
sudo pip install git+https://github.com/Ericmas001/HVAC-IR-Control
sudo pip install pigpio
sudo pip install flask
```

I use flask and a little webservice to control it, but that is not madatory.
My IR led is on GPIO 23 (Pin 16). 

app.py
```
#!/usr/bin/python

import logging 
from logging.handlers import RotatingFileHandler
from datetime import datetime, time
from flask import Flask, request, jsonify
from flask.json import JSONEncoder
import handler

class CustomJSONEncoder(JSONEncoder):
    def default(self, obj):
        if isinstance(obj, datetime):
            return obj.isoformat()
        if isinstance(obj, time):
            return obj.isoformat()
        
        return obj.__dict__

class MyFlask(Flask):
    def make_response(self, rv):
        if hasattr(rv, 'response') and rv.response is None:
            return super(MyFlask, self).make_response(rv)
        if hasattr(rv, 'new_url') and rv.new_url is not None:
            return super(MyFlask, self).make_response(rv)
        return super(MyFlask, self).make_response(jsonify(rv))

app = MyFlask(__name__)
app.json_encoder = CustomJSONEncoder

@app.route('/config/', methods=['GET'])
def list_config():
    return handler.list_config(app)

@app.route('/hvac/off/', methods=['GET'])
def power_off():
    return handler.power_off(app)

@app.route('/hvac/on/', methods=['GET'])
def power_on():
    return handler.send_last_command(app)

@app.route('/hvac/command/', methods=['POST'])
def send_command():
    return handler.send_command(app, request.json)
[... 170 lines cut ...]
    "climate_mode": "Hot",
    "end_time": -1,
    "fan_mode": "Auto",
    "isee_mode": "ISeeOn",
    "powerful_mode": "PowerfulOff",
    "start_time": -1,
    "temperature": 21,
    "vanne_horizontal_mode": "NotSet",
    "vanne_vertical_mode": "Auto"
  },
  "decoded_command": {
    "area_mode": 128,
    "climate_mode": 8,
    "end_time": null,
    "fan_mode": 128,
    "isee_mode": 64,
    "powerful": 0,
    "start_time": null,
    "temperature": 21,
    "vanne_horizontal_mode": 0,
    "vanne_vertical_mode": 64
  },
  "success": true
}
```

gouldner, 2018-08-16:
Great instructions for Raspberry!  Thanks.  If I run your Flask code as root it works fine.  But if I run it as pi I get an error. and nothing is sent.  (Code still returns success but I tested and signal isn't sent)
2018-08-16 20:55:53 gpioSetMode: pigpio uninitialised, call gpioInitialise()
2018-08-16 20:55:53 gpioWaveClear: pigpio uninitialised, call gpioInitialise()
Error in clearing wave!
127.0.0.1 - - [16/Aug/2018 20:55:53] "GET /hvac/off/ HTTP/1.1" 200 -

my user "pi" is in the group gpio and I am updated to the latest version of stretch.  Any idea how to make this work running as pi instead of having a web service running as root?

lumanga, 2019-06-12:
@Ericmas001

Dear Sir,
I have Raspberry pi 3 with Raspbian LITE OS.
I have Broadlink Mini R3 connected to my WiFi. I have IP and MAC of Broadlink inserted into the .py file.
I have installed "sudo pip3 install broadlink" and all the files of this project.
If I do in terminal:
pi@CyberPiLITE:$ python3 SendHVACCmdToRM2.py 
I receive:
`
26022601703a0e2a0e2a0e0d0e0d0e0d0e2a0e0d0e0d0e2a0e2a0e0d0e2a0e0d0e0d0e2a0e2a0e0d0e2a0e2a0e0d0e0d0e2a0e0d0e0d0e2a0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e2a0e2a0e0d0e0d0e0d0e2a0e0d0e2a0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e2a0e2a0e0d0e0d0e2a0e2a0e0d0e0d0e0d0e0d0e2a0e2a0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e0d0e2a0e2a0e0d0e2a0e0d0e2a0e0d0e0d0f0032020d`

**so python works**!

Even if I do:
pi@CyberPiLITE: $ **python3 SendHVACCmdToRM2.py -h**
I receive:
`
usage: SendHVACCmdToRM2.py [-h] [-t HVAC_TEMPERATURE] [-p]
                           [-c HVAC_CLIMATE_CODE] [-Vv HVAC_VANNE_V_CODE]
                           [-F HVAC_FAN_MODE] [--version]

Short sample python HVAC_IR command sender to Broadlink RM2 Mini
optional arguments:
  -h, --help            show this help message and exit
  -t HVAC_TEMPERATURE, --temperature HVAC_TEMPERATURE
                        Set HVAC Temperature in Celcius, Ex: 21
  -p, --power           HVAC Power, default = Power Off
  -c HVAC_CLIMATE_CODE, --climate HVAC_CLIMATE_CODE
                        Define Climate Code : C=Cold, H=HOT
  -Vv HVAC_VANNE_V_CODE, --vanne_vertical HVAC_VANNE_V_CODE
                        Define Vertical Vanne Mode : A=Automatic, S=Swing,
                        B=Bottom, T:Top
  -F HVAC_FAN_MODE, --fan HVAC_FAN_MODE
                        Define Fan speed : A=Automatic, L=Low, M=Middle,
                        F=Fast, S=Silent
`
**So I don't understand the command to use after to let my MSD-FD25 work**

For example POWER ON, COLD, 23°C, van AUTO, fan AUTO??

jgroezin, 2020-10-08:
Thanks for the code and the explanation.  Running your python on RPI.  Still figuring this out a bit but defintely controls my HVAC.