r45635/HVAC-IR-Control #24: Any way to get "durations" out of HVACDemo

newzealandpaul, 2018-05-30:
I am trying to use the HVACDemo program to output "duration" data that I can use to control my Panasonic HVAC unit using a Broadlink device. The duration format simple shows microseconds of time spent on and time spent off.

It might look like this  

+3500 -1750 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435

The +3500 means turn on for 3500 and -1750 means off for 1750 microseconds.

I tried adding this code

void IRsend::mark(int time) {
  Serial.println("+" + (String)time);
...
void IRsend::space(int time) {
  Serial.println("-" + (String)time);
...

But the durations I outputted do not seem to work. Does anyone know if I did this correctly, or if I am missing a step? 

Many thanks!

bt4wang, 2018-09-21:
This HVAC-IR-Control gives you a function to control HVAC instead of using those raw duration code. If HVAC-IR-Control doesn't works for your HVAC, please dump your remote's IR raw code and paste them here. Somebody may help then.

r45635, 2019-01-01:
Hello @newzealandpaul,
For BroadLink usage, I did an adaptation for my own purpose at home. Please have a look here: https://github.com/r45635/www-HVAC-Remote-Control I've modified the core of the algo in order to generate a RAW frame suitable for a IR Broadlink. I did not port all functions only what i needed. Such IR broadlink devices have a powerful IR range compare to what i've been able to build up to now.
cheers,
-r45635

closed 2019-01-01 by r45635