r45635/HVAC-IR-Control #26: Tested with Mitsubishi lines

gouldner, 2018-08-20:
This code works with MSZ-FE12NA.
Not working with MSY-D36NA which is weird because I have MSZ-GE and MSZ-FE and MSY-D remotes and they all work on each other.

I wonder why a MSZ-GE remote works with my MSY-D36NA but this code doesn't?

gouldner, 2018-09-21:
@bt4wang I would be happy to.  I have a Raspberry Pi with a IR receiver on it.  I don't have a copy of AnalysisIR yet.  If I purchase is it a relatively easy process to dump the raw codes?

bt4wang, 2018-09-23:
you don‘t need a paid version of AnalysIR, a free sketch for arduino is available here: https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/.

you need an arduino board and an ir receiver to run this sketch, or may be you can mod this sketch and run it on rasp pi platform.

gouldner, 2018-09-27:
Thanks, I'll look into it and try to get the raw codes.

Ericmas001, 2019-01-01:
I think we just found out in #2. MSY has a special second burst that includes a byte change. So "twin" packets are not 100% identical

gouldner, 2019-01-01:
That sound promising.  Is there a fix for this that I can test?

gouldner, 2019-01-06:
OK I tried the suggestion in #2 but it doesn't work.
I altered the file you suggested and I also altered mitsubishi.py in HVAC-IR-Control/python/hvac_ircontrol since I am using the Flash Web approach on a raspberry pi from #20 

I altered line 229 from 
data = [0x23, 0xCB, 0x26, 0x01, 0x00, 0x20,
             0x08, 0x06, 0x30, 0x45, 0x67, 0x00,
             0x00, 0x00, 0x10, 0x00, 0x00, 0x1F]
to
data = [0x23, 0xCB, 0x26, 0x01, 0x00, 0x20,
             0x08, 0x06, 0x30, 0x45, 0x67, 0x00,
             0x00, 0x00, 0x20, 0x00, 0x00, 0x1F]

That still works on MSZ-FE12NA still Not working with MSY-D36NA.
NOTE: since byte 14 was 0x10 in this file and not 0x00 like in the other code I also tried 0x30 but this doesn't work either.

If I am using the python code did I make the correct change?

bt4wang, 2019-02-14:
It will help a lot if you can get an IR dump of MSY-D36NA's remote controller.

Klinenator, 2020-09-02:
I have a mitsubishi MSZ-FH12NA it looks like maybe different codes than FE12NA

I can grab the raw codes from lirc how can i decode them so i can use them with this?

bt4wang, 2020-09-02:
maybe you could upload raw code here.

Klinenator, 2020-09-02:
Here is what i got when i pressed temp up

https://pastebin.com/PttQQfcW

bt4wang, 2020-09-02:
Good. I'll try to figure out your IR code ASAP.

Klinenator, 2020-09-02:
Thank you i really appreciate it

gouldner, 2020-09-02:
That would be great!  I have purchased some IR hardware and the AnalysIR software but never got around to figuring out how to hook it all up and scan the remote.  I would love to be able to control this AC.  I have 5 Mitsubishi split AC and this is the only one I can't control with this software.

Klinenator, 2020-09-02:
You have a default of 

0x23, 0xCB, 0x26, 0x01, 0x00, 0x20,
0x08, 0x06, 0x30, 0x45, 0x67, 0x00,
0x00, 0x00, 0x10, 0x00, 0x00, 0x1F

I believe for the MSZ-FH12NA it is
0xC4, 0xD3, 0x64, 0x80, 0x00, 0x04, 
0x1A, 0x50, 0x60, 0x02, 0x00, 0x00, 
0x00, 0x00, 0x0B, 0x40, 0x00, 0xFF

When i pressed temp up 0x50 in spot 9 changed to 0x5B

Klinenator, 2020-09-03:
Sorry i think those are for changing the temperature, disregard that

gouldner, 2020-09-03:
Mitsubishi split ac system remotes send all settings on any change.   So temp change, set cool, set hot, fan speed, etc all send all settings.

You can verify this by doing the following.  Point the remote away from AC or move to another room.  Change several settings on the remote, things you can notice happening like change fan speed from 1 to 3, change vent from down to up etc.  Now return to the AC and press temp change.   Notice that the AC changes fan speed, vent direction etc with just the press of the temp change.

Klinenator, 2020-09-03:
I have all that figured out i have the binary code for

off
temp77
temp78
temp79
temp80

so far

gouldner, 2020-09-03:
Is temp77-80 showing enough of a pattern to do any temp?

Does it look like you can map the current logic to the codes you are finding?  I have tried changing the code somewhat but my attempts have always failed.  I was just using a standard unix irc to read and it wasn't giving me meaningful output.  

I then purchased some raspberry ir boards and the Analysis IR software but never got around to learning how to use it correctly.  I will be thrilled if you and bt4wang can modify this base to get it working!  If not I have to use all the stuff I bought to get it done :)

Klinenator, 2020-09-03:
Tomorrow I need to map more temperatures to fine a pattern. I think I need 8-10 temperatures

bt4wang, 2020-09-03:
you need to grab some more raw code so we can determine what the exact
frame difine is. Your IR code was attached here.

Klinenator, 2020-09-04:
These are the 16th and 17th bytes for temperature going from 75-81 any help in figuring out the ryhme or reason to this would be nice

00000000 01000000-75

00000000 01010000-76

00010000 01010000-77

00010000 01000000-78

00000100 01000000-79

00000100 01010000-80

01000000 00000101-81

bt4wang, 2020-09-04:
Could you upload all the raw dump of each temperature?

Klinenator, 2020-09-04:
https://pastebin.com/aJ91Y6D0

bt4wang, 2020-09-04:
Have you tried to dump raw code for mode change (heat/cool/auto/dry) and
fan speed change?

Klinenator, 2020-09-04:
No but I can

bt4wang, 2020-09-04:
> 
> No but I can
> […](#)

I'm afriad that you have your raw code miss converted. Your machine use BYTE7(0 based, aka BYTE0,...,BYTE17) to set temperature. I've done an ugly transform from BYTE7 to temperature, but I think there should be an pretty way to do that. I'm not from the US so I‘m not familiar with Fahrenheit, maybe someone else can improve this. Please see the attached pdf document for more details.

You could grab more raw code if you want. Then I may help to figure out other defines, such as mode/fan/vanne/timer. 

[IR CODE.pdf](https://github.com/r45635/HVAC-IR-Control/files/5173258/IR.CODE.pdf)